PrecisionGazeMouse/PrecisionGazeMouse #80: Are multiple monitor setups supported?

skyeg3, 2022-03-26:
I use 4 big monitors in a 2x2 setup for my architectural work. The monitors are angled in about 10 degrees each to face me. Will the software work across all 4 screens?

nash8350, 2022-03-26:
No, Tobii Eye Tracker only works on a single screen. I've not tested it with multiple screens using other devices. If you try it, let me know how it goes.

skyeg3, 2022-03-28:
Well I use the bottom left screen 90% of the time. Do you think the software would work on that one screen but still allow me to use the mouse on the other screens?

nash8350, 2022-05-15:
Looks like Tobii might support this use case. I haven't been able it to test it with PrecisionGazeMouse since I only have 1 screen though https://help.tobii.com/hc/en-us/articles/209529429-Can-I-use-multiple-screens-

skyeg3, 2022-05-16:
Yeah looks like it. Thank you. I may just have to experiment with it.

nash8350, 2022-08-01:
Let us know if it works! Closing in the meantime.

closed 2022-08-01 by nash8350