PrecisionGazeMouse/PrecisionGazeMouse #81: .Net 4.5.2 End-of-life

joevano, 2022-04-07:
.net 4.5.2 will be EOL on April 16, 2022. This means no security updates.

Microsoft advises .NET developers to migrate their apps to at least .NET Framework 4.6.2 or later before April 26, 2022, to keep receiving technical support and security updates.

.NET Framework 4.6.2 (shipped almost six years ago) and .NET Framework 4.8 (shipped three years ago) are both compatible in-place replacements and stable runtimes already "broadly deployed to hundreds of millions of computers via Windows Update (WU)."

nash8350, 2022-05-15:
Thanks I'll look into updating the version when I make a new release. Not sure when that will be though.

nash8350, 2022-05-16:
I updated it to 4.7.2 in release 1.17. Let me know if you have any issues with it.

closed 2022-05-16 by nash8350