PrecisionGazeMouse/PrecisionGazeMouse #83: Low vision Q before buying hardware.

chaosdrop, 2022-05-14:
Would this work in low-vision cases where someone [myself] can only read the screen if my head is within 8~10inches of the screen and only have one working eye. My right eye mostly follows my left, but it has no vision and can wonder a little.

nash8350, 2022-05-15:
@chaosdrop Tobii says "The operating distance of a Tobii Eye Tracker is 50 — 95 cm (20 - 37")." https://help.tobii.com/hc/en-us/articles/210250305-Positioning-in-front-of-an-eye-tracker. You might want to try Talon or Optikey which have verbal mouse controls.

closed 2022-05-15 by nash8350

stickynoteme, 2022-05-15:
Do you have a link for "Talon" It's too generic of a word and I'm not finding  it using google.

shervinemami, 2022-05-15:
Talon is at https://talonvoice.com/

Talon includes both speech recognition and eye tracking, it runs very fast and is actively maintained and has a very active chat group on Slack. It supports Windows, Mac and Linux and is free cost but partially closed-source.

I haven't tried Optikey since I'm on Linux.