PrecisionGazeMouse/PrecisionGazeMouse #88: GazeMouse not working with newest Tobii Driver | starting from version 4.143.0.19463

frosk1, 2022-08-09:
I have bought the tobii eye tracker 5 and tested it on windows 11 and windwos 10.
Both versions did not work with Gaze Mouse and the current driver  Tobii.EyeTracker5.Offline.Installer_4.153.0.22216.

No I am lost and sitting on hardware, that I can not use unfortunately.

Gazemouse 1.17 is not starting, due to error message.

Will there be a patch coming soon to make Gazemouse work again @nash8350 ?

Because in the current state this software is not usable for any user, which makes this a serious issue!

frosk1, 2022-08-09:
(I am not an experienced windows user, I would switch to windows only for wanting to use this software here...)

I realized the following:

Installing only the new driver without the Tobii Experience app from the microsoft app store. leads to the following.

Setup:

1. run the the Tobii.EyeTracker5.Offline.Installer_4.153.0.2221.exe
2. do not install the Tobii Experience app from the microsoft store

--> Now win OS has the driver installed, check in devie manager
--> An app called Tobii Experience Software For WIndows (EyeTrcker5)   appears in aps & features somehow

Now I can start Precision Gaze Mouse, but

**Status  No EyeX connection**

see screenshot here:

[https://ibb.co/C107x6D](url)

frosk1, 2022-08-09:
OKay now I am super confused. 

Nothing changed and I also installed the Tobii Experience App again.

And suddenly precision gaze mouse cann identify it:

[image]

device manager:
[image]

apps installed:
[image]

driver events:

**Device deleted**
Device SWD\IS5LEYETRACKER5\SWC&USBVID_2104&PID_0313&MI_007&8BABCCD&0&0000 was deleted.

Class Guid: {5c4c3332-344d-483c-8739-259e934c9cc8}

**Device configured(null)**

Device SWD\IS5LEYETRACKER5\SWC&USBVID_2104&PID_0313&MI_007&8babccd&0&0000 was configured.

Driver Name: null
Class Guid: {00000000-0000-0000-0000-000000000000}
Driver Date: 
Driver Version: 
Driver Provider: 
Driver Section: 
Driver Rank: 0x0
Matching Device Id: 
Outranked Drivers: 
Device Updated: false
Parent Device: USB\VID_2104&PID_0313&MI_00\7&8babccd&0&0000

**Device configured (oem13.inf)**

Device SWD\IS5LEYETRACKER5\SWC&USBVID_2104&PID_0313&MI_007&8babccd&0&0000 was configured.

Driver Name: oem13.inf
Class Guid: {5c4c3332-344d-483c-8739-259e934c9cc8}
Driver Date: 06/21/2022
Driver Version: 1.106.0.22200
Driver Provider: Tobii AB
Driver Section: TobiiDriver_Install.NT
Driver Rank: 0xFF0000
Matching Device Id: SWC\IS5LEYETRACKER5
Outranked Drivers: 
Device Updated: false
Parent Device: USB\VID_2104&PID_0313&MI_00\7&8babccd&0&0000

frosk1, 2022-08-09:
---------------------------------------------

nash8350, 2023-01-31:
Ok thanks maybe you needed to reboot or something? Since you can't reproduce it I'll close this one out.

closed 2023-01-31 by nash8350

MarioVzi, 2023-04-17:
Hello!  i get the same thing with windows 10 and the same driver.  I have uninstalled both several times.  how did you finally solve the problem?  
maybe someone can provide me with an older driver?  otherwise I will unfortunately have to send my tobii back.  but i really need it...

frosk1, 2023-04-21:
hey @MarioVzi I can not really reproduce it. But there is other software you can use like: https://talonvoice.com/