PrecisionGazeMouse/PrecisionGazeMouse #92: Mac?  I'm not a good enough programmer for it but I'd sponsor someone else doing it.  Who to talk to to get started?

Detnator-Neo, 2022-11-08:
Hi there.  This looks awesome.  Macs and iDevices need this.  In fact I'm amazed Apple hasn't come up with something as promising as this looks.

I have minor disabilities that make using a mouse sometimes challenging, but I almost exclusively need to use Apple devices for various reasons.  

I'd like to see this working on a Mac as soon as possible.  Who can I talk to about financially or otherwise sponsoring such an effort?

I realize I could just hire someone familiar enough with both Mac and Windows programming to port this, but I'd welcome the opportunity to collaborate with the author of this in the process.

Can we get in touch and discuss please?

Thanks!

MousyBusiness, 2023-01-31:
@Detnator-Neo I'm thinking to start a project on this front (out of necessity for a Mac eye mouse). I too would appreciate some sessions with the original author to understand the various challenges he faced.

nash8350, 2023-01-31:
Hi I am the author. What would you like to know? Also, have you tried using Talon voice https://talonvoice.com/docs/?

MousyBusiness, 2023-01-31:
Hi @nash8350 thanks for the response. I appreciate your work in getting an opensourced version of an eye mouse available on Windows and would like to create a similar solution for Mac users. I will first get a Tobii 5 and set up with Talon to see what kind of results are possible.

1. Am I correct in understanding that your application takes outputs from an eye-tracking hardware and transforms it into a meaningful mouse solution?
2. Is the reason you have developed for Windows because these hardware solutions only have Windows drivers (that you're aware of). 
3. What major pitfalls did you encounter developing this software (Lessons hard learned)?

nash8350, 2023-02-03:
1. Yep along with head rotation for precision movements
2. Yes Tobii only offers Windows drivers. Talon has reverse engineered the drivers for other OSes.
3. Eye tracking alone is too jittery to use as a mouse. You need to pair it with a smoothing filter, and something more stable like head tracking for finer movements.