PrecisionGazeMouse/PrecisionGazeMouse #93: Switching to Talon Voice

nash8350, 2023-01-31:
Hi everyone this project has been hugely helpful in helping me maintain my career over the past several years. My latest job requires me to use Mac which is not supported by PGM. As a result, I'd either have to port and rewrite it as a cross platform solution or switch to another one. I started this project because there were no other options available at the time, but I've always hoped that other commercial and open source mousing solutions will adopt the functionality and maintain it.

I'm happy to say that [Talon Voice](https://talonvoice.com/) is now a good solution with cross-platform support and compatible functionality. It also has native support for both eye and head tracking without using third party software. I recommend PGM users to try using Talon. For example, to enable movement on key press and click on key, the following talon file will work:

```
key(f3:down):
  user.mouse_toggle_control_mouse(1)
key(f3:up):
  user.mouse_toggle_control_mouse(0)
  mouse_click()
```

I've since switched all my desktops to Talon. I don't have a need for this project anymore, so I will stop contributing to it. However, it is open source so I will let the community take if forward if you desire.

nash8350, 2023-01-31:
Please let me know if someone is interested in becoming a maintainer for this project.

shervinemami, 2023-01-31:
Thanks for all your great work & commitment to PGM over the years! ❤️ Talon isn't completely open-source but it works amazingly well and is updated very often and has a very active community, so I think it's a good replacement for anyone that doesn't specifically need an open-source solution👌

AshtonMacSaylor, 2023-06-14:
I was so excited on finding this, only to realize you have stopped supporting it. I've tried to get Talon up and running, but it's a beast, and I have little to no use for the voice commands. 

Would you be able to post a tutorial for raw newbies about how to set up Talon to mimic exactly what PGM does (and nothing more)?

nash8350, 2023-10-08:
@AshtonMacSaylor you could just disable speech recognition if you're not using it. You'll want to setup eye tracking and then create a talon file in your user directory following this guide https://talonvoice.com/docs/#talon-files. Paste in the configuration in the example above, save it, then use the F3 key to enable and disable eye tracking.

NoSTs, 2026-06-27:
Hello, since the one click installer is not working due to the website/server being down I made a build of the source code of 1.7.
Its the only way to get this program.
I can not add it to this repo so I made a fork. Its under releases there.
https://github.com/NoSTs/PrecisionGazeMouseReleaseBuild/releases/tag/myfirsttag

I dont want to be a maintainer.