PrecisionGazeMouse/PrecisionGazeMouse #94: Unhandled Domain Exception: Configuration system failed to initialize

edmun777, 2023-03-04:
I keep getting this message every time I tried to run the program.  I used the Control panel to reset to an earlier version of the software the first time, which corrected the issue for awhile.  Then when it started happening again, this option was no longer available.  I uninstalled the software and tried for a fresh install, but I am now getting the same error every time I try to install it and the program will not install.

nash8350, 2023-04-08:
There might be a corrupted configuration file. Check for a config file by opening explorer and entering `%AppData%` in the location bar. Then browse to find the PGM folder and delete it. Then reinstall it to start fresh.

edmun777, 2023-07-31:
Deleting and reinstalling the program does not seem to fix this.

edmun777, 2023-07-31:
The %AppData% search does not currently show any folder for PGM, and when I try to install it fresh, the Unhandled Domain Exception pops up.

nash8350, 2023-10-14:
Sorry try %localappdata%