PrecisionGazeMouse/PrecisionGazeMouse #95: Flagged as malware

TopoRopo, 2023-03-20:
[image]

Bitdefender seems to flag this as a threat, likely due to machine learning that is also used to spread BadRabbit ransomware.

gitihobo, 2023-04-04:
I want to use it but until that gets fixed we cant be sure right?

nash8350, 2023-10-14:
You might want to try uninstalling the application and deleting the AppData folder, then reinstall it. I don't have bit defender, but I just uploaded a zip of the whole PGM website including install files to Kaspersky and found no match. 

[image]
https://opentip.kaspersky.com/B42BFEFD785510D58E6C37AC5F7696D37DE06CF1E576FF0198B25AD78875A322/results?tab=upload